matrix-io/matrix-creator-quickstart #4: Failed to start Program FPGA and SAM3 MCU

mukulseth, 2016-07-24:
Hi,

After following the getting started instructions, I get the above message in the console logs, and the following systemctl --

>> systemctl status matrix-creator-firmware.service 
● matrix-creator-firmware.service - Program FPGA and SAM3 MCU
   Loaded: loaded (/lib/systemd/system/matrix-creator-firmware.service; enabled)
   Active: failed (Result: exit-code) since Sun 2016-07-24 20:48:37 UTC; 2min 8s ago
  Process: 352 ExecStart=/usr/share/admobilize/matrix-creator/creator-init.bash (code=exited, status=1/FAILURE)
 Main PID: 352 (code=exited, status=1/FAILURE)

Does anyone know what's going on? I'm using a Pi Model B+, and using the Raspbian Jessie Lite May 27th 2016 image

arhuaco, 2016-07-25:
@mukulseth, can you restart the raspberry and report back?

mukulseth, 2016-07-25:
Tried without any luck... then I tried removing the packages, auto-remove, and then reinstalling again.. same result.

arhuaco, 2016-07-25:
What happens if you run both
/usr/share/admobilize/matrix-creator/fpga-program.bash
/usr/share/admobilize/matrix-creator/creator-init.bash
?

mukulseth, 2016-07-25:
attaching the error output.. 
[matrix-jtag-error.txt](https://github.com/matrix-io/matrix-creator-quickstart/files/382135/matrix-jtag-error.txt)

preview --

XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

No JTAG Chain found
**** Could not program FPGA

AND

Open On-Chip Debugger 0.10.0-dev-00283-g7223cae-dirty (2016-07-15-08:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO config: tck = 17, tms = 4, tdi = 22, tdo = 27
BCM2835 GPIO config: trst = 18
trst_only separate trst_push_pull
adapter speed: 512 kHz
Warn : em357.bs: nonstandard IR value
Warn : Specify TAP 'sam3n.cpu.cpu' by name, not number 3
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG only mode enabled (specify swclk and swdio gpio to add SWD mode)
Info : clock speed 512 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: em357.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Could not initialize the debug port
Error: Target not examined yet

**** Could not program SAM3 MCU

arhuaco, 2016-07-26:
Hi @mukulseth.

We tried the getting started steps and we managed to program the FPGA.

https://github.com/matrix-io/matrix-creator-quickstart/wiki/getting-started
The steps have been simplified in the last few days.

We tried on the Model B+ and the mic array and the leds work.

We could not program the MCU and we will probably need to modify openocd to program the MCU correctly.

arhuaco, 2016-07-26:
Update: All the instructions work on the Raspberry V2 and V3.

mukulseth, 2016-07-26:
Any recommendations to get it working on a Model 1 B+ based on the error log I attached? I have tried following those instructions exactly multiple times.

arhuaco, 2016-07-27:
Something we know is that the FPGA works with the Raspberry that you have. Now, we need to know why it doesn't work for you. What power supply are you using? Let's make sure it's delivering enough power for both boards.

mukulseth, 2016-07-27:
I'm using a 5V 2A adapter with a high quality USB cable. Haven't had any problem running the more powerful Pi 2 and Pi 3 with this adapter. Do I have to power the Matrix separately?

arhuaco, 2016-07-27:
@mukulseth That's a great adapter. Is there a possibility to try the Creator with a Pi 2 or a Pi 3? You don't have to power the MATRIX Creator separately.

eljuguetero, 2016-07-27:
@mukulseth we recommend a 2A power adapter with very low ripple, both FPGA and LEDs are power hungry.

mukulseth, 2016-07-28:
I tried on a Pi 3 with the same result unfortunately.. Is there something wrong at a hardware level possibly?

arhuaco, 2016-08-08:
Dear Mukul:  As per our communication via email, we will ship you your unit back to us for diagnosis as we are pretty certain it is not the hardware, as yours was the only unit from the entire first batch that appear to not be working as intended. We look forward to getting you to create like others soon enough.  Thank you.

bensechrist, 2016-08-30:
Getting a similar error but only for programming the SAM3 MCU

Open On-Chip Debugger 0.10.0-dev-00283-g7223cae-dirty (2016-07-15-08:36)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO config: tck = 17, tms = 4, tdi = 22, tdo = 27
BCM2835 GPIO config: trst = 18
trst_only separate trst_push_pull
adapter speed: 512 kHz
Warn : em357.bs: nonstandard IR value
Warn : Specify TAP 'sam3n.cpu.cpu' by name, not number 3
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG only mode enabled (specify swclk and swdio gpio to add SWD mode)
Info : clock speed 512 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: em357.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Could not initialize the debug port
Error: Target not examined yet

**** Could not program SAM3 MCU

kdpatino, 2016-08-30:
Hey @bensechrist,

Which model of the RPi did you use? Right now we've tested RPi 2 and 3.
Could you give us more details about your installation steps that led up to this issue?

bensechrist, 2016-08-30:
Using a RPi Model B+

I followed the getting started instructions

eljuguetero, 2016-08-31:
@bensechrist 

We found a bug in the MCU programming script that only happens in the RPi Model B+
`sudo  aptitude update  && sudo aptitude upgrade`

bensechrist, 2016-08-31:
Perfect. That seemed to fix it. Thanks

closed 2016-08-31 by eljuguetero