matrix-io/matrix-creator-quickstart #6: broken links

arhuaco, 2016-08-19:
After an update to the name of wiki pages there are a few broken links in the wiki.

Once pages receive a name it's hard to change it without updating the rest of the wiki.

Too sleepy to fix and finishing a document.

arhuaco, 2016-08-19:
Wiki URIs  don't usually look like real titles. They follow simple names like: how-to-update-the-wiki.

brianofrokk3r, 2016-08-19:
@arhuaco let's provide meaningful names so we don't have to change them again. We don't control the URIs, they are auto-generated. I will go through and fix, I adjusted only 3.

arhuaco, 2016-08-19:
Hi there Brian. The issue is that the order that you put in the titles might not be stable on time. It's more important to keep these pages reachable for search engines than providing names that make sense for a limited amount of time. I am awake now, I can help.

closed 2016-08-19 by arhuaco