matrix-io/matrix-creator-quickstart #8: Node.js Documentation

chalkers, 2016-08-30:
Hi there,

The examples in the matrix os repo are for remote access only and for only a narrow set of functionality. How do I use Node.js directly to get data on the Pi itself?

Also, how do I get access to the NFC?

Regards
Andrew

eighteyes, 2016-08-31:
Andrew,

MatrixOS is not a node module, so it is not a typical use of Node.js. MatrixOS is intended to run on the Pi and provide data to a distributed infrastructure and make data  available via webhooks, csv exports or dashboards. 

@eljuguetero can speak towards how to access the NFC.

thanks!
Sean

arhuaco, 2016-08-31:
On the other hand, you'll be able to access more devices today via MALOS (what is documented in the wiki). Nothing prevents you to run Node.js from the Raspberry Pi.

But as @eighteyes said, the ultimate goal is to use the device via MatrixOS.

Only microphones and GPIO support will be missing.

I'll update this soon when the new devices are ready (UV driver and pressure driver).

arhuaco, 2016-08-31:
"The examples in the matrix os repo are for remote access only and for only a narrow set of functionality. How do I use Node.js directly to get data on the Pi itself?"

Andrew,  I just pushed the new MALOS package to the Debian repository so you can "apt-get update && apt-get upgrade" to get the most recent version.

I also updated the wiki page with information of how to run Node.js from the Raspberry itself.

https://github.com/matrix-io/matrix-creator-quickstart/wiki/4.-MALOS

Let me know if it helps.

LengZai, 2016-09-02:
Hi @arhuaco

I'm very interested by using the mic array for multiple vocal command for home automation like Google Home and Alexa...
As Matrix Creator Mic array is not recognized as a audio record device by Pi3. It will be great to be able to get a DSP process to get 1 was file from the 8 microphones dedicated to work on the human voice frequency...
You mentioned "Only microphones and GPIO support will be missing."
Which mean we won't be able to record voice from mic array through JS? Only through cpp?

Hard to maintain through cpp only as need to compile for every testing.

Thanks hope you can give us some ideas.

arhuaco, 2016-09-02:
Hi @LengZai, the current support is C++ only. But we are planning JS support. I think @eljuguetero is more qualified to answer your question.

LengZai, 2016-09-02:
Thanks for your quick reply... 
JS or Python will  be great

My aim is to use Matrix Creator with this project...
https://github.com/alexylem/jarvis
You should give a try.. the application is in English...
I was thinking to use your micarray_demo binary to exploit the 8 microphones to generate an unique raw file (instead of 8) then convert in wav file with Sox to send it to Google or Bing API... for STT part.

robopsi, 2016-09-18:
hi, this may sound silly as i am a programming noob, but is there a program that can let me change what the led arrays do? i wanted a colour sweep,, continually phasing from one colour to the next.

nebiljabari, 2016-09-25:
@arhuaco In the link you mention upper

> I also updated the wiki page with information of how to run Node.js from the Raspberry itself.
> https://github.com/matrix-io/matrix-creator-quickstart/wiki/4.-MALOS

You should include those command lines like indicated in the [matrix-io/matrix-creator-malos
README.md](https://github.com/matrix-io/matrix-creator-malos) : 
`- malos > /dev/null 2>&1 & ` // start MALOS process in background
`- pkill -9 malos` // kill MALOS process

Otherwise the node script run but nothing happen

arhuaco, 2016-09-27:
@nebiljabari You are right. I'll update this documentation but the best solution is to have MALOS start on reboot. We should be adding startup scripts very soon and you should get them by updating the Debian package. (Not yet, soon).

brianofrokk3r, 2016-10-06:
@arhuaco this updated?

arhuaco, 2016-10-06:
Yeah. I just updated the README as well. MALOS runs as a service now.

brianofrokk3r, 2016-10-06:
Thanks @arhuaco!

closed 2016-10-06 by brianofrokk3r