matrix-io/matrix-creator-quickstart #9: 3. HAL Everloop LED Array And IMU

nebiljabari, 2016-09-24:
Ive check and re-check many times and I think I followed correctly the steps but I was unable to get the `demos/everloop_demo` work. No led get lighted

When I run `demos/arc_demo` I get random light _(blue, red, green, white)_ on random upper LED and almost all the LED on the bottom part are off.
When I do `ctrl c` after `demos/arc_demo` the LED lighten stay on until I reboot

Im on a RaspberryPi 3 Model B _(ssh remote control with Macbook Air and wifi)_ - All packages are update - The small SAM3 MCU LED flash every 1s

eljuguetero, 2016-09-24:
@nebiljabari random light patterns are usually associated with bad power supply unit. MATRIX Creator needs  2A, 5.0V, low ripple PSU.

nebiljabari, 2016-09-24:
@eljuguetero I have a official RaspberryPi power supply (perchase with a kit). The sticker indicate:
Output: +5.1V - 2.5A

You still think the PSU is guilty ? If yes, any advice(s) ?

nebiljabari, 2016-09-25:
@eljuguetero "You was right", now my Matrix look like a real ufo !
Thx to [Flynnprgmmr](http://community.matrix.one/t/power-issues-with-matrix-and-raspberry-pi3/61) for his post on the forum

I know we should but who read the safety guide ?

> WARNINGS
> This product shall only be connected to an external power supply rated at 5V DC, with a minimum amperage of 2A

I think it really deserve a note in the [wiki Getting Started page](https://github.com/matrix-io/matrix-creator-quickstart/wiki/1.-Home)

closed 2016-09-25 by nebiljabari