matrix-io/matrix-creator-quickstart #10: sam3mcu programming - echo 18 > /sys/class/gpio/export

nebiljabari, 2016-09-25:
When I follow the echo list you provide (as root and/or sudo) I get for the first line :
```
echo 18 > /sys/class/gpio/export
bash: echo: writing error : Peripheral or occupied ressource
```
My OS language is in French maybe my translation is imprecise.
`bash: echo: erreur d'écriture : Périphérique ou ressource occupé`

The others lines don't send any error and when I enter : `sudo openocd -f cfg/sam3s.cfg` 
I get almost the same "Example output" as you :

> Flashing succeeded if you get the line "wrote 38620 bytes from file blob/ch.bin to flash bank 0 at offset 0x00000000 in 1.256279s (30.021 KiB/s)"

My output :
`wrote 33604 bytes from file blob/ch.bin to flash bank 0 at offset 0x00000000 in 1.135267s (28.906 KiB/s)`

Should I get concerned about the echo 18 line or it's normal (acceptable) ?

kdpatino, 2016-09-26:
Hi @nebiljabari, regarding echo 18 error, is just a warning, that you can be ignored, it means GPIO 18 is alerady registered in the sys file systems. So as you can see, openocd had written the MCU flash memory. 

BR,
Kevin

nebiljabari, 2016-09-26:
Thank you for your responses @kdpatino and I will follow your advice

closed 2016-09-26 by nebiljabari