matrix-io/matrix-creator-quickstart #15: Failed to flash the MCU. RGB-LEDs and Mic on the FPGA work fine.

NilzOS, 2016-10-06:
After going through the quickstart guide nearly everything worked fine. But I was not able to get sensor data. I was trying to flash the sam3 controller but no chance. This is what happens while trying to flash the mcu on the Pi2 and Pi3.

root@raspberrypi:/usr/share/admobilize/matrix-creator# ./sam3-program.bash

cat /tmp/sam3-program.log
Open On-Chip Debugger 0.10.0-dev-00286-g83b68a2-dirty (2016-08-30-15:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO config: tck = 17, tms = 4, tdi = 22, tdo = 27
BCM2835 GPIO config: trst = 18
trst_only separate trst_push_pull
adapter speed: 512 kHz
Warn : em357.bs: nonstandard IR value
Warn : Specify TAP 'sam3n.cpu.cpu' by name, not number 3
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG only mode enabled (specify swclk and swdio gpio to add SWD mode)
Info : clock speed 512 kHz
Info : JTAG tap: em357.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: em357.bs tap/device found: 0x069aa62b (mfg: 0x315 (Ember Corporation), part: 0x69aa, ver: 0x0)
Info : JTAG tap: xc6sxl4.fpga.fpga tap/device found: 0x24000093 (mfg: 0x049 (Xilinx), part: 0x4000, ver: 0x2)
Info : JTAG tap: sam3n.cpu.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Error: xc6sxl4.fpga.fpga: IR capture error; saw 0x35 not 0x11
Warn : Bypassing JTAG setup events due to errors
Info : sam3n.cpu: hardware has 6 breakpoints, 4 watchpoints
sam3n.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00402520 psp: 0x20000800
#0 : at91sam3 at 0x00400000, size 0x00020000, buswidth 0, chipwidth 0
	#  0: 0x00000000 (0x4000 16kB) not protected
	#  1: 0x00004000 (0x4000 16kB) not protected
	#  2: 0x00008000 (0x4000 16kB) not protected
	#  3: 0x0000c000 (0x4000 16kB) not protected
	#  4: 0x00010000 (0x4000 16kB) not protected
	#  5: 0x00014000 (0x4000 16kB) not protected
	#  6: 0x00018000 (0x4000 16kB) not protected
	#  7: 0x0001c000 (0x4000 16kB) not protected
flash 'at91sam3' found at 0x00400000
Error: Invalid ACK (0) in DAP response
Error: Failed to read memory at 0x400e0a0c
Error: SAM3: Error performing Erase & Write page @ phys address 0x00403c00
Error: Invalid ACK (6) in DAP response
Error: Failed to write memory at 0x00404600
Error: SAM3: Failed to write (buffer) page at phys address 0x00404500
Error: error writing to flash at address 0x00400000 at offset 0x00000000

I bet it´s a hardware malfunction.
And yes, the Pi war powered by a 5V 2A power supply.

Any ideas guys ?

NilzOS, 2016-10-06:
Hey guys,
after struggeling for days, it was a 5 minute job.
Login as root, then
killall malos
cd /usr/share/admobilize/matrix-creator
./sam3-program.bash
It was finally done. Don´t ask why. I guess the malos thead interfered with the flashing operation. Perhaps it was the case that systemd mashed things up when it was starting services.

So everything is fine now.
So long and thanks for all the fish.

Nilz

arhuaco, 2016-10-06:
@NilzOS 

This is a pending issue we have. We need to do better error recovery for this case. @eljuguetero and I know that we need to solve this very soon.

https://github.com/matrix-io/matrix-creator-init/issues/4

closed 2016-10-06 by arhuaco