matrix-io/matrix-creator-quickstart #19: malos-eye

dajor, 2017-01-10:
I follow your instruction but i have a problem when i start 
node test_query_demographics.js i only get: 
**Sending pings every 3 seconds**

I can not find anything in a log file what is the error

luisnomad, 2017-01-23:
No answer...

dajor, 2017-01-24:
No - i could not make it working

brianofrokk3r, 2017-01-24:
Hey @dajor @luisnomad  did you follow the install process? Please check first that `ps aux | grep "malos"` services are running. If they are, please kill them.

Upgrade all your services
``` 
sudo apt-get update
sudo apt-get upgrade
sudo pkill -9 malos
sudo pkill -9 malos_eye
malos_eye
```

Install the examples
```
# Clone repository.
git clone https://github.com/matrix-io/malos-eye.git

# Fetch protocol-buffers repository.
git submodule init
git submodule update

# Setup examples.
cd examples
npm install
```

Run the examples
```
node query_demographics.js
```

Please run through this and let us know if you continue having issues. Once this is resolved this ticket will be moved to the proper repository.

brianofrokk3r, 2017-01-24:
@luisnomad @dajor 
please also provide OS version

arleneF, 2017-05-17:
Same issue here! I download the OS on May 15th (don't know the version number). 
And I also tried the method @brianofrokk3r suggested, but still nothing happened.

ps: after I kill the malos and` ps aux | grep "malos"` gave me `pi        3842  0.0  0.2   4276  1852 pts/0    S+   11:48   0:00 grep --color=auto malos
` (which I think this is fine)

Need help to make detection work!

Predhin, 2018-10-30:
is there support for Windows OS ?
Thanks,