matrix-io/matrix-creator-quickstart #28: Installation does not work with new  Raspbian Stretch

mingfang, 2017-08-25:
Fails here https://github.com/matrix-io/matrix-creator-quickstart/blob/master/install.sh#L24

```
sudo apt-get -y --force-yes install malos-eye matrix-creator-malos matrix-creator-openocd matrix-creator-init;
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 matrix-creator-malos : Depends: libzmq3 (>= 3.2.3+dfsg) but it is not installable
 matrix-creator-openocd : Depends: libjim0.75 (>= 0.73) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

vineet-sinha, 2017-09-07:
I am having the same problems above.

If it helps I noticed that I do have `libjim-dev`, `libjim0.76`, and `libzmq3-dev`.

mingfang, 2017-09-07:
As a temporary workaround, you can add the old jessie repo.
```echo "deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi"|sudo tee /etc/apt/sources.list.d/jessie.list```

vineet-sinha, 2017-09-07:
That works. Thanks!

llangar, 2017-09-16:
I have an issue at this point:
cp: no se puede efectuar `stat' sobre '/etc/lirc/hardware.conf': No existe el fichero o el directorio
dpkg: error al procesar el paquete matrix-creator-init (--configure):
 el subproceso instalado el script post-installation devolvió el código de salida de error 1
Se encontraron errores al procesar:
 matrix-creator-init
Any Sugestion?

vineet-sinha, 2017-09-17:
@llangar - I had the same problem, but was able to get it to work by first running `sudo touch /etc/lirc/hardware.conf`.

llangar, 2017-09-18:
Thanks It works!!