matrix-io/matrix-creator-quickstart #30: The matrix-creator repository installed by the script does not work.

jtickle, 2017-09-16:
From Raspbian Stretch on a Raspberry Pi 3, the mirror "deb http://packages.matrix.one/matrix-creator/ ./" will not be loaded by 'apt update' or 'apt-get update' because it does not contain a Release file at that path.

If I attempt to go directly in a web browser (on a different computer) to http://packages.matrix.one/matrix-creator/Release, I get this:

<Error>
<Code>AccessDenied</Code>
<Message>Access denied.</Message>
<Details>
Anonymous users does not have storage.objects.get access to packages.matrix.one/matrix-creator/Release.
</Details>
</Error>

I get a similar error trying to browse anywhere else on the packages.matrix.one domain.

gordinh, 2017-09-25:
Hi @jtickle we are working on packages to support MOS at Raspbian Stretch. So for now it's normal to not be updating. We are expecting to update this week.

mikevador02, 2017-10-31:
Hello,

I have the same problem today 31 oct 2017. 
On clean raspberry strech lite
My have to install old realase : jessie lite?

1)typing:
echo "deb http://packages.matrix.one/matrix-creator/ ./" | sudo tee --append /etc/apt/sources.list

2)typing:
sudo apt-get update

output is:
....
Ign:8 http://packages.matrix.one/matrix-creator ./ Translation-en_GB
Ign:9 http://packages.matrix.one/matrix-creator ./ Translation-en
Ign:8 http://packages.matrix.one/matrix-creator ./ Translation-en_GB
Ign:9 http://packages.matrix.one/matrix-creator ./ Translation-en
Fetched 11.8 MB in 9s (1,244 kB/s)
Reading package lists... Done
W: The repository 'http://packages.matrix.one/matrix-creator ./ Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Can we help me?

Thanksin advance.

gordinh, 2017-11-01:
Hi @mikevador02 please try with
```
curl https://apt.matrix.one/doc/apt-key.gpg | sudo apt-key add -
echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list
```
It should work with both Jessie and Stretch, we are updating our new packages.

mikevador02, 2017-11-03:
Hello,

Thanks for your fast response.
So i have re started with stretch (i was tested with jessie with succes only with the HAL demo, after i had much problems to use the mike array (noise on the sound) ).

So now it's ok with the update up to ***:

curl https://apt.matrix.one/doc/apt-key.gpg | sudo apt-key add -

echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list

# Update packages and install
sudo apt-get update
sudo apt-get upgrade

# Installation
***sudo apt install libmatrixio-creator-hal libmatrixio-creator-hal-dev
***up to this os OK.

A)After with this command on strecth to install demo (or later the matrix-init...) i have (with jessie was ok):

sudo apt-get install cmake g++ git libfftw3-dev wiringpi matrixio-creator-init libgflags-dev

The error message about hash:
Get:1 https://apt.matrix.one/raspbian stretch/main armhf matrixio-openocd armhf 1.0.1 [840 kB]
Err:1 https://apt.matrix.one/raspbian stretch/main armhf matrixio-openocd armhf 1.0.1
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:f3c24501968b70e593a903d902bc78996558991c8028213f77dd8c405266b336
   - SHA1:51779de7fd3b5a6f5f2f010c18868c9c88108819 [weak]
   - MD5Sum:8a09dde5339dbe96145fbf13e48634ee [weak]
   - Filesize:840168 [weak]
  Hashes of received file:
   - SHA256:1519160871216df290417055d56ca335d7ec6c903cdc6ed9e4c63b594d77ca07
   - SHA1:580b081f57683048b66c8ca287a4371cde2fb614 [weak]
   - MD5Sum:c0cdfce1f1b34c142e0e29899df3b171 [weak]
   - Filesize:839928 [weak]
  Last modification reported: Wed, 01 Nov 2017 13:28:02 +0000
Fetched 840 kB in 1s (567 kB/s)
Unable to correct missing packages.
E: Failed to fetch https://apt.matrix.one/raspbian/pool/m/ma/matrixio-openocd_raspbian-stretch-1.0.1_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:f3c24501968b70e593a903d902bc78996558991c8028213f77dd8c405266b336
    - SHA1:51779de7fd3b5a6f5f2f010c18868c9c88108819 [weak]
    - MD5Sum:8a09dde5339dbe96145fbf13e48634ee [weak]
    - Filesize:840168 [weak]
   Hashes of received file:
    - SHA256:1519160871216df290417055d56ca335d7ec6c903cdc6ed9e4c63b594d77ca07
    - SHA1:580b081f57683048b66c8ca287a4371cde2fb614 [weak]
    - MD5Sum:c0cdfce1f1b34c142e0e29899df3b171 [weak]
    - Filesize:839928 [weak]
   Last modification reported: Wed, 01 Nov 2017 13:28:02 +0000
E: Aborting install.

So i can't continue.

B)
And if i want use array mikes later 

sudo apt-get install matrix-creator-init cmake g++ git 

sudo apt-get install matrix-creator-init cmake g++ git
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package matrix-creator-init is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  matrixio-creator-init

E: Package 'matrix-creator-init' has no installation candidate

So i tried to add the old (worked on jessie but not on stretch)
# Add repo and key
curl https://apt.matrix.one/doc/apt-key.gpg | sudo apt-key add -
echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list

And i go back to the firt problem why i have open this forum thread.

Thanks for your help.

jaron42, 2018-04-22:
I am having a similar problem after
`echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list`
I get 
`The repository 'https://apt.matrix.one/raspian stretch Release' does not have a Release file.`
with the more specific error being a 404 after looking for
`https://apt.matrix.one/raspian stretch/main armhf Packages`
Any suggestions? @mikevador02 did you ever get stretch working?
Thanks for any suggestions.

chaseadam, 2021-11-11:
closing loop. Can confirm this works for buster (newer than stetch) 32 bit OS versions:

```
 echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list
```