matrix-io/matrix-creator-quickstart #31: Error when start matrix os

llangar, 2017-10-11:
NODE_ENV=dev node index.js
Environment Selected: dev
Loading… device
device Loading… device > gpio +0ms
device Loading… device > heartbeat +12ms
device Loading… device > maintain +10ms
device Loading… device > malos +4ms
device Loading… device > network +2ms
device Loading… device > port +135ms
device Loading… device > sensor +3ms
device Loading… device > service +3ms
device Loading… device > system +5ms
device Loading… device > wifi +2ms
device Loading… device > driver > accelerometer +4ms
device Loading… device > driver > altitude +3ms
device Loading… device > driver > detection +4ms
device Loading… device > driver > gesture +4ms
device Loading… device > driver > gpio +3ms
device Loading… device > driver > gyroscope +5ms
device Loading… device > driver > humidity +3ms
device Loading… device > driver > ir +4ms
device Loading… device > driver > led +3ms
device Loading… device > driver > magnetometer +7ms
device Loading… device > driver > mic +2ms
device Loading… device > driver > pressure +11ms
device Loading… device > driver > recognition +5ms
device Loading… device > driver > temperature +283ms
device Loading… device > driver > uv +3ms
device Loading… device > driver > voice +3ms
device Loading… device > driver > zigbee +6ms
device Init Cmds Registered:> 15 accelerometer face demographics palm thumb-up fist pinch gpio gyroscope humidity magnetometer pressure recognition temperature uv +9ms
Loading… event
matrix Loading… event > app +4ms
matrix Loading… event > device +6ms
matrix Loading… event > firebase +4ms
matrix Loading… event > heartbeat +3ms
matrix Loading… event > sensors +3ms
matrix Loading… event > server +4ms
matrix Loading… event > service +4ms
matrix Loading… event > token +4ms
matrix Loading… event > util +2ms
matrix Loading… event > ves +2ms
Loading… service
matrix Loading… service > application +7ms
matrix Loading… service > auth +341ms
matrix Loading… service > component +3ms
matrix Loading… service > cypher +2ms
matrix Loading… service > firebase +4ms
matrix Loading… service > grpc +781ms
grpc GRPC:> malos : heartbeat +416ms
grpc GRPC:> vision : recognition +99ms
grpc GRPC:> vision : recognition_service +72ms
grpc GRPC:> vision : vision +28ms
grpc GRPC:> vision : vision_service +40ms
matrix Loading… service > lifecycle +1ms
matrix Loading… service > manager +3ms
matrix Loading… service > protobuf +14ms
proto malos : driver.proto +6ms
proto malos : hal.proto +89ms
proto malos : heartbeat.proto +7ms
proto malos : zwave_commands.proto +113ms
proto vision : recognition.proto +65ms
proto vision : recognition_service.proto +1ms
proto vision : vision.proto +16ms
proto vision : vision_service.proto +12ms
matrix Loading… service > stream +23ms
matrix Loading… service > token +7ms
matrix Loading… service > track +59ms
matrix Loading… service > tracking +3ms
matrix Loading… service > ves +3ms
matrix Loading… service > zeromq +3ms
matrix Debug: ,-engine,-Component*,-led,-gatt,-bleno,-bt-characteristic,-hci +56ms
matrix ====== config ===vvv +0ms
matrix { fakeApp: undefined,
matrix path:
matrix { root: ‘/home/pi/matrix-os/’,
matrix db:
matrix { device: ‘/home/pi/matrix-os/db/device.db’,
matrix config: ‘/home/pi/matrix-os/db/config.db’,
matrix user: ‘/home/pi/matrix-os/db/user.db’,
matrix service: ‘/home/pi/matrix-os/db/service.db’,
matrix pending: ‘/home/pi/matrix-os/db/pending.db’ },
matrix pendingFiles: ‘/home/pi/matrix-os/public/pending_files’,
matrix update: ‘/tmp/matrix-update/’,
matrix backup: ‘/tmp/matrix-backup/’,
matrix apps: ‘/home/pi/matrix-os/apps’,
matrix protos: ‘/home/pi/matrix-os/proto’,
matrix splash: ‘public/splash’,
matrix appLog: ‘/home/pi/matrix-os/apps/app.log’ },
matrix jwt: { secret: undefined },
matrix version: ‘0.15.0’,
matrix local: {},
matrix heartbeatInterval: 10000,
matrix spaceLimit: 5000000,
matrix splashInterval: 30,
matrix sensorRefresh: 2500,
matrix sensorSockets: true,
matrix socketCheckDelay: 1000,
matrix socketRefreshTimeout: 10,
matrix registrationUUID: ‘b1a6752152eb4d36e13e357d7c225466’,
matrix configurationUUID: ‘b1a6752152eb4d36e13e357d7c225467’,
matrix envs:
matrix { deviceSecret: ‘d1b5d16afeb91a03fc3ec0d22936bd23df3789ea53a6ec00d7b65aba2d102d53f1a26c7379f96360ed1229b948’,
matrix deviceId: ‘dfc4d3b15135’,
matrix nodeEnv: ‘dev’ },
matrix components: {},
matrix paths: { root: ‘/home/pi/matrix-os/config’ } }
+0ms
matrix ENV: dev API: https://dev-api.admobilize.com MXSS: http://dev-mxss.admobilize.com +89ms
matrix ENV: dev API: https://dev-api.admobilize.com MXSS: http://dev-mxss.admobilize.com +0ms
device device init: heartbeat +147ms
device driver init: accelerometer +48ms
device driver init: altitude +1ms
device driver init: detection +47ms
device driver init: gesture +1ms
device driver init: gpio +0ms
device driver init: gyroscope +1ms
device driver init: humidity +0ms
device driver init: ir +1ms
zeromq [zmq][registerComponent> led +1ms
device driver init: led +9ms
device driver init: magnetometer +2ms
device driver init: mic +1ms
device driver init: pressure +1ms
device driver init: recognition +33ms
device driver init: temperature +1ms
device driver init: uv +1ms
device driver init: voice +0ms
device driver init: zigbee +9ms
matrix Loader stopped +2ms
Not using device data from db, using MATRIX_DEVICE_ID and MATRIX_DEVICE_SECRET instead!
matrix event listener: app +149ms
matrix event listener: device +1ms
matrix event listener: heartbeat +1ms
matrix event listener: sensors +1ms
matrix event listener: server +1ms
matrix event listener: service +1ms
matrix event listener: token +1ms
matrix event listener: util +0ms
matrix event listener: ves +2ms
Starting as device: dfc4d3b15135
sdk Offline setup successful +5ms
sdk Checking API server… +6ms
sdk secret-> d1b5d16afeb91a03fc3ec0d22936bd23df3789ea53a6ec00d7b65aba2d102d53f1a26c7379f96360ed1229b948 +474ms
sdk [post] params: { url: ‘https://dev-api.admobilize.com/v2/device/token’,
form:
{ device_id: ‘dfc4d3b15135’,
device_secret: ‘d1b5d16afeb91a03fc3ec0d22936bd23df3789ea53a6ec00d7b65aba2d102d53f1a26c7379f96360ed1229b948’,
jwt_token: true },
json: true } +5ms
sdk [post] body: { status: ‘error’, status_code: 400, error: ‘Device not found.’ } +329ms
sdk Device Error { status: ‘error’, status_code: 400, error: ‘Device not found.’ } +8ms
API Error: 400 - Device not found.
Unable to setup online MOS, something went wrong (400) (Device Token Retrieval Error)
Bad Matrix Initialization Device Token Retrieval Error
sdk Loader stopped +7ms
sdk Loader stopped +2s
sdk DESTROYING +8ms
sdk Firebase wasn’t initialized +8ms
Cleanup complete…

gordinh, 2017-10-25:
This device (`device_id`) isn't registered. You need to use a registered device. You can register by CLI with `matrix register device`.

closed 2017-10-25 by gordinh

llangar, 2017-10-25:
It is correct this device is no longer registered, I had to register another but the result is the same.
The Id I have now is d5f9c6fcac6a

gordinh, 2017-10-25:
You are still receiving `Device_not_found` ? The environment you registered is the same from the environment you are using in MOS?

llangar, 2017-10-25:
The environment that I am using is RC same as in matrix-cli or in the phone, I have mxss connection in the application of the telephone but it does not respond to any action does not turn on the LEDs, in version 0.14.1 I worked correctly.

gordinh, 2017-10-25:
When your MOS is initializing the LEDs turn on?

llangar, 2017-10-25:
No, with the previous version when running mos lights on and then shut down until you ran any application.
Again thank you very much for replying

gordinh, 2017-10-25:
Can you send me your device_secret and then i can test here on my device? You can send to lucas.morais@admobilize.com
I will check it now and answer here

llangar, 2017-10-25:
I have already sent to you

AlexandreSi, 2018-07-13:
Hi,
I have the same issue here. I registered my device and trying to run the HelloWorld example or the first green led App but the leds do not answer. I installed the matrix-creator-hal to run the demos and they work. But I am unable to make the apps run on my matrix voice.
When I saw this post (mos lights are not turned on when initializing either), I tried to run in dev mode and I got the same output.
@gordinh were you able to fix the issue of @llangar?